SmartEVSE/smartevse #1: OCPP Implementation

goRaspy, 2015-11-16:
Hi,

Your works is great ! Did you already thought about implementation of "open charge point protocol" for smart EVSE ?

Regards,

wido, 2015-12-18:
No, not really. I think nobody started any work on this.

It might be really simple that a additional controller simply sets a pin to high and if that pin is low no charging is allowed.

tho-wa, 2016-05-09:
Yes that's a great idea, then the SmartEVSE could also be used for public charging.
There are multiple billing service providers that are compatible to OCCP.

wido, 2018-11-05:
I think that #9 would help with this. I'm closing this one right now as when #9 is implemented people could add their external auth to SmartEVSE.

closed 2018-11-05 by wido