SmartEVSE/smartevse #2: Control charge current by solar power

tho-wa, 2015-12-25:
Is it possible to change the smart mode this way that it would be possible to measure power of solar system/photovoltaic with the sensor box and control the charge current based on the pv power output?

wido, 2015-12-25:
In theorie it is possible. The sensor box just sends the values of how many Amps each phase is using. If you fake the input into the EVSE with something that comes from your PV array you could do that.

tho-wa, 2015-12-30:
The sensor can be put on the PV AC cable after inverter or before main meter of the house so the PV power can be messurred. There is only an additional option needed in CLI  and extension of the code, so that the charging current is calculated by PV power.

tho-wa, 2016-02-01:
Does the sensorbox measure energy flow in both direction, or only in one direction?

mstegen, 2016-02-23:
The sensorbox can not detect the direction of flow of current. So it can not tell the difference between a load or your PV system generating power.

The RS485 connection between module and sensorbox uses the same protocol SMA is using on their inverters. So if you have a SMA inverter (and RS485 piggyback), you should be able to read the generated Power by modifying the code.

ljfaragher, 2016-10-09:
Tho-wa
Did you get smart evse controled by excess solar pv. Thus us exactly what I want to do.

jackdwest, 2017-04-06:
I am using the Smart to control charging using excess PV.  My system is off grid and I measure the DC current from the array using CT4 input and a shunt instead of CT.  This was fairly straight forward to implement.  It would be very similar to implement with an AC array too using CT3.

closed 2018-03-11 by tho-wa