SmartEVSE/smartevse #7: Minimum max mains current is too high

xalerius, 2018-04-24:
The lowest mains current that can be set is currently 25A. Is there a reason for this? In Sweden 16A mains is quite common.

wido, 2018-04-24:
In the Netherlands (where it was developed) 25A is the default, so that's where it probably came from

xalerius, 2018-04-25:
Ok that makes sense, well then I guess it should be possible to change to a lower value. I tried to make that change myself yesterday, but when I try to run my code I get garbage characters on the display (and in the serial interface). If I enter the menu (by pressing the middle button) I get normal text, but as when I exit the menu I get garbage instead of "READY TO CHARGE". I have tried to just build a version without any changes and get the same behaviour so it seems there is something wrong with my environment. Anyone else had this problem? I have a v2 board, tried to build under both windows and linux, same result.

lordprize, 2018-04-26:
I have changed this for a single SmartEVSE at the request of a friend just a few days ago. If more people need it, perhaps the change can be implemented in the official software. Perhaps a value even lower than 16A would be OK (e.g. 11A - one more than the minimal MaxCurrent).

Elektrofix-OL, 2018-05-13:
I‘ve the same problem. The mains current need to be less the 16A.
I try to built a neu firmware, but the compiler needs someone files:
„xc.h“
„stdio.h“
„stdlib.h“
„math.h“
and I didn‘t find them. Where is my mistake?

mstegen, 2018-05-16:
Sorry for the delay. I just changed the lowest MaxMains setting to 10A in version 2.03.
Had to change the CLI limits as well.

lordprize, 2018-05-16:
Perfect - thank you, Michael!

Elektrofix-OL, 2018-05-16:
Thanks, my charging station at work is now completely finished. The charging station fuse will no longer trip. [https://www.goingelectric.de/forum/ladeequipment/ein-ladepilz-entsteht-t31485.html#p736339](url)

There ist a mistake while the smart mode: wenn L2 or L3 be on limit, but L1 has enough space, the EVSE doesn't switch on. That isn't nice, because the EV get no power.
Perhaps it is possible to start changing without looking on L2 and L3 for th first 5 secons?
There are a lot of EV witch use only L1 in Germany. EV whitch use L1-L3 are not so often.

mstegen, 2018-05-17:
SmartEVSE needs atleast 6A of room to start charging. (as that's the lowest charge current it can set) If any of the phases is higher then MaxMains-6A, then it will not start. That's implemented for security.
Unfortunately there is no way for the EVSE to know if the connected EV will only charge at one phase, or if all phases will be used.

You could always change the code yourself, or disconnect L2 and L3, on both the contactor and sensorbox, so that it will charge at one phase only.

mstegen, 2018-05-17:
I have tried to reproduce the problem, but did not see any issues.
I measured the PWM duty-cycle of the CP signal, and it's 26,6% as it should be at 16A.

A possible explanation could be that you compiled the sourcecode yourself. I noticed that some new compiler versions will complain a lot. There might be wrong code generated at the SetCurrent function.
The default duty cycle is then set to 10% which is 6A.

Please try to flash the hex file in /version2/SmartEVSE2.X/dist/default/production/