SmartEVSE/smartevse #9: Use IO for authorization

ErnieV, 2018-07-11:
Getting a signal on I/O port opens a window of 1 minute to start charging program, otherwise charging is not allowed. Gives user possibility to implement any form of authorization like hidden button, NFC/RFID reader or remote control by webserver.
As optional feature (menu item).

arjannv, 2018-11-30:
I like the idea and would like to help programming this. 
Maybe it's a good idea to, for instance set I/O: 
I/O 1: High
I/O 2: Read for access control
I/O 3: High when charging is possible

The access control circuit then needs to provide a potential free contact to I/O 1 and 2 for access control. In this way, there is no hassle with grounding, overload protection etc.

I/O 3 can be used to create a switching contact near the Smart EVSE. With this switching contact a Red/Green LED can be powered near the connector.

A few questions:
- Which ports are used for I/O 1, 2 and 3?
- Is there a way to get a High signal without using the precious I/O 1 port for this?
- Can I use the Timer variable (mSec) for the 1 minute of access? This needs to incorporate a function for the overflow of-course

mstegen, 2019-02-20:
Software version 2.04+

I have now added support for authorization with a (momentary) switch on IO2 (SW on hardware 2.1)
One pulse to ground, and access is enabled, another pulse will disable the EVSE (and also stop charging)

You can enable this feature in the (LCD) menu of the SmartEVSE, the feature is called ACCESS.
When set to Disable, a (push)button on IO2 will still function as a STOP button, to be able to stop the EV from charging.
You will have 15 seconds to remove the cable. After 15 seconds the charging will continue, and cable is locked again.

On hardware version 2.1, IO1 is now used to drive a 12V led. It will fade in and out when charging.

IO3 will be used to connect to a residual current monitor device, to detect AC/DC fault currents.
This will come very soon in a future update.

arjannv, 2019-02-20:
Thanks for the update!

I thought so already while browsing the menu of my SmartEVSE :)
Also spotted the new IO notes. 

How is this switch connected? Is IO2 reading for a 12V input?

Can this also be used to schedule the charging? Waiting for the off-peak hours? 

I've got my SmartEVSE up and running now. Only need to test the ACCESS part with a Raspberry Pi reading for RFID tags and familiar Bluetooth devices.

mstegen, 2019-02-20:
Schedule the charging is tricky, as there is no real time clock in the SmartEVSE.
So it does not know what time it is.
The 16Mhz xtal could be used as time reference, but not sure what the time drift would be.
I think it's better to schedule the charging from the EV. (if your car supports it)

Plugging in the car and then at a specific time enabling access to the SmartEVSE (with the Raspberry Pi) would give you scheduled charging.
Btw, That's a nice idea, ref the familiar Bluetooth devices!

arjannv, 2019-02-21:
Ah yes, timing with the Raspberry Pi I meant :)

Regarding the Bluetooth idea:
I came across this thread: https://www.raspberrypi.org/forums/viewtopic.php?f=37&t=47466
It shows how to connect to the (known) Bluetooth device and read the signal strength. In this way, you can authorize the charging only when the Bluetooth device is very near.

spikerm, 2019-03-25:
I managed to get a fingerprint sensor working as acces device :)

question when the car is disconnected and acces is enabled in the software does the cable keeps locked until right acces key is given ?
so the cable keeps locked in the charger

mstegen, 2019-03-26:
Nice that you got a fingerprint sensor to work!

The current software always unlocks the cable from the socket, once the charging is stopped.
Maybe an idea for a future feature?

pascalsaul, 2020-07-17:
@spikerm @mstegen Could you share and update the documentation which components we need to buy and how to wire to add the new access (un)lock funtion?

mstegen, 2020-07-26:
You'll need a stand alone RFID reader, one that works with mifare cards (13.56 Mhz), and has a relay output. And can be learned to recognize your charge card.
Usually these relay outputs will switch on for a short period of time, and then switch off again (pulsed output).
Connect the relay to the SW and GND pins on the SmartESVE, and set the SWITCH option in the menu to ACCESS B (for button).

Now when you swipe your card , the smartESVE can be switched from ACCESS DENIED, to READY TO CHARGE. and vice versa.

pascalsaul, 2020-12-26:
I got it working with a rewritten ESPeasy module but I didn't like the implementation. In the meantime I use a Sonoff Basic with Tasmota where I cut off the 230V and just have a stupid relay. The Sonoff is configured as a Pulse button for 0,5 second. This combination I'm using together with OVMS to stop charging at 80% now through Home-Assistant.

pascalsaul, 2021-01-14:
@ErnieV I think this issue can be closed now :)