SmartEVSE/smartevse #14: Loadbalancing not working as expected with mixed 1 and 3 phase cars

barteld, 2018-12-15:
The current loadbalancing algorithm only takes into account the current, and not the number of phases the cars can charge at. This leads to cars waiting to get charged while enough power is available. 
To further explain: I have a ZOE (that can handle up to 3x64A) and have ordered a Niro. The Niro has a single-phase charger and will be charged through a transformer (https://www.ratio.nl/nl/catalog/e-mobility/ev-transformer-laadstation/ev-transformer-laadstation/37300/groups/g+c+a+nr+view).

Now when the Niro is charging the ZOE has to wait until it has completely finished as the maximum current of my charging group is 3X16A. This is a bit of a waste as the ZOE can charge just fine at 1X16A while the Niro uses 2X16A. All in all this means that the total charging time for both cars goes up from 10hrs to 15 hrs from almost emprty.

I understand that this is not an easy one to fix as the PWM signal only allows for the current to be set, and cannot be used to instruct the car to only use a specific phase (as far as I know, didn't read the whole specification).

However there is a workaround possible if using fixed cables. In that case we can use a second contactor wired up (with a 12v relais) to the LOCK output. We can reconfigure this output to shut of 2 out of three phases when in loadbalancing mode. In this case the Niro uses for instance L1 and L2 to charge while the ZOE uses only L3 when the Niro is charging and L1, L2 and L3 when it's the only one.

This will work for up to three cars. The cleanest possible solution would be for evse to be able to monitor all phases and share info about which phases are used but in my opinion evse should be kept as simple as possible. The main target for evse is home charging but it should be able to handle the various chargers as a lot of cars go on the market now with just single phase support.

I have created a version of evse that does this, however it is not to be used as it is not tested and it's my first go at embedded programming. I will share it in the comments but regard it as a POI (proof of idea) rather than something that is to be used.

barteld, 2018-12-15:
Here's the code changes:
https://github.com/SmartEVSE/smartevse/compare/master...barteld:Phase_based_loadbalancing?expand=1

And a link to the repo:
https://github.com/barteld/smartevse/tree/Phase_based_loadbalancing

ankowolf, 2019-01-04:
This is exactly what I aim at doing. In my case it will be a car with a 1x32a OBC (Opel Ampera E) and car with a 3x24a OBC (Tesla). When the Opel is charging, the Tesla can charge on 2 phases. When the Opel is not charging the Tesla can charge on 3 phases.

Have the same idea with the additional contactor.

But have the same issue that the first module must take into account the load on L1 where the second module must take into account the load on L2 + L3 (and optionally L1).

As I didn't want to lean how to program the module, I decided to implement my own master on a RPI Zero with an RS485 interface and use two modules, both in Slave mode.

Getting there, but the result is not stable. For example, both slaves switch from Charging State to State A when they receive a broadcast message from the master that advertises 7 amps current for either one of the slaves. Same with 11 amps. As if they don't like 7 and 11.

Rings a bell?

evhaandel, 2019-01-11:
Hi @barteld ,

You're mentioning that you will use the SmartEVSE in combination with EV Transformer Charging Station from Ratio Electric to charge a Kia e-Niro. When will you receive your e-Niro? I'm very interested in your experiences with this charging combination. Can you contact me via my Github profile? 

Thanks,
Emiel

arwooldridge, 2019-08-31:
@barteld If I understand you correctly this will not work, the three phase to single phase transformer cannot be fed from only two of the three phases, if so it would either produce too high a voltage or too low depending which phase was missing. In any case the three phase to single phase transformer arrangement ( wye in wye out) is not fully balanced and no transformer only arrangement can be balanced, only an inverter solution. Why not just limit the charging current from inside the EV's.
Depending on the specification of the 3phase to single phase transformer you could possibly take both single phase supplies from that ( again limiting the current in each EV to suit). Beware though that the 3Ph to 1Ph transformer setup. apart from not balanced, is also not very efficient due to winding losses and magnetic losses. Actually a difficult problem to solve neatly unless you eliminate the 3ph to 1ph transformer setupcompletely, and charge direct off 2 phases for Niro and one phase for Zoe, perhaps that is what you are trying to do?

AlbertHakvoort, 2019-12-17:
> Had a little coding issue in my RPI based slave. Escaping of characters was not done properly. Fixed that and now it seems stable.

Hi Anko, would you  like to share the code of the modbus master script?