SmartEVSE/smartevse #18: Measure kWh charged

HeNRGi, 2019-07-22:
It would be great to have a measurement of kWh measured in the SmartEVSE itself. 

Could it be as simple as multiplying current times interval duration and add this (for all phases) to a kWh counter?
Accuracy may vary, based on measurement sources, e.g. 
LOW - current setting sent to EV and seconds timer used as input
MEDIUM - current measured via a (secondary) CT sensor box and seconds timer as input
HIGH - same as previous , but intervals are now measured via timestamps obtained vian another input means (RS486 modbus?)

N.B. In e.g. The Netherlands, MID compliance is NOT needed for kWh measurement data for tax purposes (contrary to what meny people seem to believe), according to the legal department of the Dutch Tax and Customs Administration.

bobosch, 2019-07-23:
Yes, this is on the todo list.

But it would not be very precise.
First, a phase detection is needed. Many cars charging with one phase, many with three...
Also the internal timer is not very precise. And if the sensorbox allows 32 A charging current, and the car supports only 16 A, this won't be recognized by the smartEVSE. (this error could be reduced by only allowing 16 A...)

A second sensorbox would be more precise, but currently it is not possible to use two sensorbox because they have a fixed address.

Supporting an electric meter would be easy - but then you can just as well take a look at the display of the electric meter.

HeNRGi, 2019-07-23:
Based on the discussion over at [Tweakers (in Dutch)](https://gathering.tweakers.net/forum/view_message/59498712), it became clear to me that kWh measurement based on the current setting sent to the EV might grossly overestimate the amount of electiricity used to charge; an EV is free to choose any current lower than the maximum it gets from the SmartEVSE, and towards the end of the charging session, almost certainly does so.

So, it seems an approach is needed that measures voltages and currents multiple times per second (say 400) similar to the way it is done in the [OpenEnergy Monitor EmonLib](https://github.com/openenergymonitor/EmonLib/blob/master/EmonLib.cpp#L66), which seems quite similar to [Application Note 939 by Microchip](http://ww1.microchip.com/downloads/en/AppNotes/00939a.pdf), the maker of the PIC microcontroller used in SmartEVSE. 

In the Netherlands, I suppose a smart meter's P1-port could be used to measure the mains power drawn of the total home, potentially freeing up the CTs and the SmartEVSE sensorbox to be retargeted towards measuring the cumulative energy spent towards charging the EV?

For Tesla EVs, another software-only approach that seems to be feasible, is using the (reverse engineered) Tesla API to repeatedly poll for the parameter charge_energy_added via the [GET charge-state endpoint of the Tesla car](https://www.teslaapi.io/vehicles/state-and-settings#charge-state). The paid TeslaFi service is based on this Tesla API and provides the info I need. In the Netherlands, the free [Jedlix service](https://jedlix.com/en/) also uses this Tesla API, but I am unsure whether they track kWh charged as well. Moreover, the Tesla API is not a properly supported API that allows 3rd parties scoped access and repeatedly polling the car seems wasteful...

pascalsaul, 2020-07-17:
There is an integration with the SDM630 meters all ready . Is it possible to use those meters to measure the charged KWH's? I've put on off those in the chargebox to measure now.  Instead of using it for the "MAINSMET" use it as "CHARGEMET?

The question is now how good the internal memory is to save the loading sessions over time and how long can the history be?

bobosch, 2020-11-25:
Implemented with commit 67f4163fd47fb145e410594e7d08d794d4de31b3 - no history, only last charge

closed 2020-11-25 by bobosch