SmartEVSE/smartevse #19: It doesn't change the power when "smart" Mode

LolDeibid, 2019-08-02:
Hi all,

I hope you can help me. I will try to explain my issue as best as i can (not native english). 

Today i finished to set up my "DIY Wallbox" with SmartEVSE and SensorBox. All fine, the sensorbox catch up all the consumption at home and i can see printed on the display when "Smart" mode is ON.

I've 230@20a at home, single phase. At "MAINS" option i setted 20a, MAX 20a and MIN 10a. I've fixed cable so i setted on this at 20a as i builded it with 4mm² cable.

The issue is that i can connect my microwave, toaster, computer and some lights, see on screen L1 - 11 (i understand 11a??), but still charging at 20a (Charging 20a (20a) at screen).

Am i failing with setup? Here you have my actual setup: 

MODE -> Smart
MAINS ->20a
MAX -> 20a
MIN-> 10a
Cable ->20a
CAL-> 01.1a (can't modify)
Access-> Disabled
RCMON->Disabled

Thanks a lot!

mstegen, 2019-08-19:
The display shows the total current measured , if it displays L1: 11, then it measures 11A on the phase.
When starting to charge, it will display the max charge current you set (20A), at the moment the car is actually drawing current the L1 measurement should increase.
When the total current measured on L1 is higher then 20A (MAINS setting) it will start to lower the charging current to the EV.
The EV will react to this within a few seconds, so that the maximum current on L1 should stabilize at 20A.

LolDeibid, 2019-08-19:
Mmmm if I understand it, i should add the sensorbox at main phase BEFORE the charger to measure all consumption, house and car, together? Actually i've connected after the car, so it "reads" only the house, as other chargers works like this.

Is it right? I will test this afternoon. 

Kind regards, 

David.

mstegen, 2019-08-19:
Yes exactly, the sensorbox and CT sensor, should be able to measure ALL current, including the car.

By doing this, it knows exactly when to start reducing the current to the car, with only one sensor.

closed 2020-08-20 by LolDeibid