SmartEVSE/smartevse #24: PV adjusted current by using Sensor Box

mcarcog, 2020-06-14:
Hi, is there already the possibility to use the Sensor box to detect the output AC power of a PV System?

So the sensors box should increase the current and not reduce.

Thanks for replying

mstegen, 2020-06-15:
The SmartEVSE expects the Sensorbox to be at the point where the electricity is feed into the house. At this location it can determine if the current is flowing to the grid, or if power is drawn from the grid. Therefore it doesn't need a separate sensorbox which only monitors the power generated by the PV system.
Btw you need the Sensorbox 2 for this to work, as the Sensorbox 1 (and 1.5) can not determine the direction of the current.

With this in place, it will try to charge the car only with solar power(in Solar mode). It does this by monitoring the measured currents, and trying to adjust the charger of the EV so that the sum of all currents is zero.

mcarcog, 2020-06-15:
Thanks much for your answer. Nice to hear, that this feature is already implemented.

closed 2020-06-15 by mcarcog