SmartEVSE/smartevse #25: Control charging current directly via Modbus

heisly, 2020-06-22:
Hello,
I like the Smart EVSE very much because of its pretty design including the display and it is a economic setup together with the RCM14, very good job!
I would like to integrate Smart EVSE within my Loxone Miniserver via existing Modbus Gateway with some existing SDM630 power meters. 
Therefore I would like the visualize the actual state (ready, vehicle connected, charging) in the Loxone App (I think this is already possible) and **control the charging current directly via modbus** (as I already have all the power and energy information and the necessary logic in loxone).

Is that already possible (maybe I haven’t seen it) or something interesting to integrate in the future?
Maybe it is possible to simply allow an override of the charging current in the Solar Mode.
I would highly appreciate that!

Otherwise: I like the Smart EVSE anyway 😉. 

Greetings, Stephan

bobosch, 2020-06-22:
Hi,
yes, SmartEVSE can use the SDM630 and you can use modbus to get some information
https://github.com/SmartEVSE/smartevse#modbus-registers
But currently you have to update the configuration to set the charging current (would be saved every time :-( )
I like the idea of overriding the charging current, but it would be only possible in normal mode. In Smart and Solar mode the charging current is updated periodically...
Regards, Robert

heisly, 2020-06-23:
Override the Charging Current in Normal Mode would also be fine. But to make sure, with the actual firmware this is not possible, right?

bobosch, 2020-06-24:
Yes, only the commands listed on https://github.com/SmartEVSE/smartevse#modbus-registers (since Version 2.10)

Fnikan, 2020-06-25:
Hi,
I have the same problem. Actually i want to send START/STOP command with modbus to SmartEvse and control the charging current too. I am thinking about adding a new register for that and modifing the code for that( The part of IO2 which is for START/STOP). But until now no idea how to add a new register to SmartEvse.
Btw thanks for this big work!

bobosch, 2020-06-25:
START/STOP charging is already possible by setting the access bit register 0xA7

bobosch, 2020-06-25:
I'll make the modbus charging current register 0xA6 writeable

heisly, 2020-06-26:
Hey bobosch, that sounds really great, thanks for that! 
BTW, the modbus Interface is working with a fixed Baudrate of 9600 Baud and a fixed Address, right? So I need to make some changes on my already existing meters to have them all on the same Modbus and bring them together with Smart EVSE within one gateway to the Modbus TCP.

bobosch, 2020-06-27:
Solved with commit 759f2c0051b062f5aac2e70332bb3b9dd5c9b2dc
Works also in Smart Mode. Don't forget to set value to 0 to let SmartEVSE control the current.
[SmartEVSE2.X.production.zip](https://github.com/SmartEVSE/smartevse/files/4840118/SmartEVSE2.X.production.zip)

bobosch, 2020-06-27:
BTW: Yes, RS485 settings and the address are hardcoded.

closed 2020-06-27 by bobosch

heisly, 2020-06-28:
That's fine, I also seen it in the EVSE.c I'm already on the same Baudrate and just have to change the address of one of my meters to reserve 1 for the Smart EVSE.