SmartEVSE/smartevse #26: Charging not working - red light on the car

bostjanlaba, 2020-07-15:
Having setup SmartEvse with fixed cable, 3 phase with RC Mon, no smart mode, no switch, 16A. 

Display says READY TO CHARGE. Plugging the cable into car shows red light on the car. SmartEvse reports nothing, but does not charge.
Only thing discovered is that for some reason pins PP and CP on the smartevse itself (without any cables attached) are shorted.

What could be the reason?

mstegen, 2020-07-15:
Make sure that you do not connect the PP wire to the charging cable. Only the CP wire (and PE) should be connected from the fixed charging cable to the SmartEVSE.
The PP connection should only be connected if a charging socket is used.

When the SmartEVSE detects a connected car, the back light of the screen will light up. It will still show "READY TO CHARGE" though. Only if the car asks for power, the contactor will close, and "CHARGING" will be displayed.

bostjanlaba, 2020-07-15:
Hi,

I have tried disconnecting PP from smartevse (though in the plug, there is still resistor between PP and PE). But the result was the same. Should I remove the pin from the plug all together? 

Does having shortage between PP and CP on smartevse is ok?

Thanks for you super quick reply!

⁣Lep dan/Best Regards

Boštjan Laba​

bostjanlaba, 2020-07-15:
Hi again,

just want to say it was a faulty cable (PP had 500 ohm instead of 680ohm)

All good now, thanks!

⁣Lep dan/Best Regards

Boštjan Laba​

mstegen, 2020-07-15:
Ah, ok thanks for letting me know.

closed 2020-07-15 by mstegen

bostjanlaba, 2020-07-15:
Hi again. Now generally chargin is working but not for long. Could be 1 hr or less but then a car reports no electricity available.

All fuses are ok, no error on the smartevse, instead of charging now it says ready to charge. Taking plug out and back into vehicle does nothing, not even initial communication is seen on the car. If i turn smartevse off and back on, station will operate normally for another short period of time as expected.

Any idea what is this? Thanks.

⁣Lep dan/Best Regards

Boštjan Laba​