SmartEVSE/smartevse #27: Config shielded with PIN

laemen, 2020-07-19:
Feature request: I have the SmartEVSE semi-outside (garage), and protected with a Raspberry Pi with RFID meter switching the SmartEVSE switch pin. 
However you can easily go info the configuration and disable the switch pin, enabling the SmartEVSE and the actuator. Shielding this with a 4 digit pin would prevent that. The SDM630 has PIN shielding for comparison.

mstegen, 2020-07-26:
Sure i can add that in a future version. But how to reset the pin in case you forget it somehow?

- Allow a PIN reset by holding a button on the front when powering on?
- Only allow PIN reset by reprogramming the controller with the FTDI cable.
- Any other idea?

laemen, 2020-07-26:
PIN reset through a button on the front would void the protection. A physical recovery would be the best, reprogramming through the FTDI would have my vote.
Couldn't find how the SDM630 recovers the PIN, maybe by buying a new one 😳

Links2004, 2020-09-20:
may this helps as base:
https://github.com/SmartEVSE/smartevse/pull/33/commits/aed6ccbbc5874c5bd411822326e6e05316b8c092

I Impediment a simple lock that can be enabled and disabled via Modbus

## Register 0xD*: Extended Configuration

Register | Access | Description | Unit | Values
--- | --- | --- | --- | ---
0xD0 | R/W | Disable Menu | | 0: Menu Enabled  / 1: Menu Disable

PR open:
https://github.com/SmartEVSE/smartevse/pull/33

bobosch, 2020-12-02:
> PIN reset through a button on the front would void the protection.

... "holding a button on the front **when powering on**" If you are able to switch the power on and off you have physical access ... IMHO the best solution.
@Links2004 yes, it should be also accessable via modbus. I will merge this commit