SmartEVSE/smartevse #28: Charging stops for no "apparent reason"

bostjanlaba, 2020-07-20:
Charging a car normally at 10,5kW for 2-3 hrs. Suddently charging stops. No error. Contactor is disconnected (off). 

Car reports that there is sudden loss in current. SmartEVSE shows Ready to charge. Disconnecting the cable and reconnecting it to the car makes no change. No communication with the car at all. 

Shutting down SmartEvse and restarting normally continues the charge.

So for some reason it looks like mains get disconnected on contactor and station goes into Ready to Charge and "freezes" there...

RC Monitor can be enabled or disabled, no difference.

Any idea why? 

Bostjan

mstegen, 2020-07-26:
Could it be that the SmartEVSE get's too hot? It will stop charging when the temperature sensor measures 65 C.
It will continue charging again if the temperature drops below 55 C.
I just checked and the current version does not show the ERROR HIGH TEMP message, just READY TO CHARGE when it get's too hot. 
I'll fix this bug in the next version.

You can monitor the temperature by connecting a FTDI cable to to SmartEVSE, and using a terminal program like putty to connect to it. Pressing enter should bring up the CLI menu

    ----------------------------- SMART EVSE -----------------------------
     v2.17 for detailed instructions, see www.smartevse.org
     Internal Temperature: 29 C
    ----------------------------------------------------------------------
    CONFIG  - Set to Fixed Cable or Type 2 Socket                - Socket
    LOCK    - Cable locking actuator type                        - Disabled
    MODE    - Set to Normal, Smart or Solar EVSE mode            - Normal
    LOADBL  - Set Load Balancing mode for 2-4 SmartEVSEs         - Disabled
    MAX     - Set MAX Charge Current for this EVSE               - 13 A
    SW      - Switch function control on pin SW                  - Disabled
    RCMON   - Residual Current Monitor on pin RCM                - Disabled
    >

bostjanlaba, 2020-07-26:
Hi

Yes, this would be quite possible. The ev charging box is outside and there is a lot of sun. Does evse get hot because of the charging or only because of hot environment?

Not surw how to cool it down though...

⁣Lep dan/Best Regards

Boštjan Laba​

mstegen, 2020-07-26:
Hi, 
If the SmartEVSE controller is next to the contactor, and this get's hot, maybe you can create a air gap between the two? (maybe rearrange them on the DIN rail?)
The controller doesn't get hot by itself, so it must be from outside sources..
-Michael

bostjanlaba, 2020-07-26:
Hi,

 

SmartEvse is between RCCB and MCBs. Contactor is one line lower but indeed just beneath SmartEvse.

 

I will try and move them further apart. I’d put in a vent but since it’s an outside box it has to be water and dust resistant :/

 

Will report back on findings.

 

Bostjan

stefan345, 2020-07-31:
@bostjanlaba I have the same issue. Out of curiosity: which car are you trying to charge?

bostjanlaba, 2020-07-31:
Hi, audi etron. Everywhere else I charge I have zero problems.

Tomorrow I am removing classic rccb and mcbs and replacing them with combo which will give me some room to have smartevse almost alone on the din rail. Hopefully it will help...

⁣Lep dan/Best Regards

Boštjan Laba​

bostjanlaba, 2022-08-08:
To whom it may concern. It's a problem with overheating. Issues are only during summer, particularly if sun is shining directly on the box. So either we cool it actively or move it into permanent shadows. We'll see.

closed 2022-08-08 by bostjanlaba

mstegen, 2022-08-09:
I've seen custom enclosures with a mesh at the backside of the enclosure (top and bottom), shielded from rain, but it allows airflow through the box.