SmartEVSE/smartevse #34: Empty if statement for PWM cycle > 92% detection

arjanmels, 2020-10-08:
The if in the first line line has a semicolon after the if, so the codeblock following will always be executed. Is this the intention?

https://github.com/SmartEVSE/smartevse/blob/fc20d524fc929fd35ff26eb7a4711927521eb577/version2/SmartEVSE2.X/EVSE.c#L2540-L2545

mstegen, 2020-10-09:
Good Catch!

No this is not intended. I'll fix it in the next release.
Thanks.

arjanmels, 2020-10-09:
I caught it because I switched my fork to c99 compiler which gives a warning for this. 

c99 compiler (clang) has much nicer error messages. However if I use the c99 lib instead of the c90 lib the size grows from ~77% to 85% of memory. It is possible to use c99 language mode with c90 library (you get some warnings in doprint.c, btu it does compile).

bobosch, 2020-11-25:
Commited d5128d51e0da15336191b8412c4f42dfb078b48f

closed 2020-11-25 by bobosch