SmartEVSE/smartevse #35: load balancing algoritme

BitterAndReal, 2020-10-16:
How is the load balancing between multiple EVSE's done at the moment if they do not measure the current each car is drawing?
Is the total available current equally shared by the EVSE's with a car connected or is there a smart algoritme which checks if a car actually uses his share of the current? 
So one car could charge with 90% of the available current and the other with 10% for example.

mstegen, 2020-10-18:
It can only determine if the car is charging or not, it will therefore for multiple EVSE's just share the max chargecurrent between the EVSE's that have a car connected and are charging. If one of the EV's is finished charging it will recalculate this value. 

The max chargecurrent is in Smart mode also determined by what's measured on the MAINS connection. If it's not exceeding the set value it will increase the chargecurrent. If it detects a higer current on one of the phases then what is set in the controller (MAINS) then it will decrease the chargecurrent.

BitterAndReal, 2020-10-19:
Thanks for the explanation. Really smart solution!
Does that mean I could have two EVSE's sharing a 25A mains connection both allowing the connected car to charge with 20A but one car is only drawing 5A for example?

mstegen, 2020-10-20:
With two cars charging, both cars will get 12.5A If one EV is only drawing 5A, the other one will still only charge with max 12.5A
until the first one is full, and stops charging, then the other one will get the full 25A.

closed 2020-11-17 by mstegen