SmartEVSE/smartevse #42: Use standard implementations for the modbus libraries

neographikal, 2020-12-17:
I ran into some implementation difficulties because SmartEVSE uses it's own modbus interpretation especially for the 0x01 & 0x02 addresses. I dove into the code and saw it builds it's own modbus message structures. Why not use libraries that are already available, conform to the industry standard and are being maintained?

bobosch, 2020-12-17:
I found no lightweight c library (program space is very small).
In current git version, address 0x01 is used as special broadcast address. With multiple electric meters and multiple SmartEVSE the communication must be efficient. Also modbus implementation on the electric meters is very different.
Address 0x02 should work normal.

neographikal, 2020-12-17:
What would you consider small? There are some libraries on github for master / slave functionality, they are a few hundred to roughly 1000 lines of code.

bobosch, 2021-01-24:
Yes, would be a possibility to check some of them, maybe add some functionality or/and adapt it to the XC8 compiler - but work is already done, it's small and has the features needed.

closed 2021-01-24 by bobosch