SmartEVSE/smartevse #43: Modbus Register 0x0* broadcast: strange implementation

neographikal, 2020-12-17:
I tried to use these functions from standard modbus libraries (Python, Arduino) but these functions are weirdly structured. Some devices do use 32bit registers, but these are quite rare. Using function code 16 to write 4 values in standard libraries result in

0x01
0x02
0x03
0x04

being written instead of 4 values at one address. This does not seem to conform to the Modbus standard. Secondly, why use broadcasting at all, why not let the master address the 4 slaves individually by writing one register with the standard write register function. This way you also have confirmation that the slave recognized the request and acts accordingly.

The docs are also quite limited on how these function should be called, just calling it does remove the communications error on the smartevse, but it seems to stall: buttons do not work any more.

bobosch, 2020-12-17:
On this modbus implementation registers are 16bit "words". I don't understand your example, as there are only 2 words. Please write down the complete code with the request statement, the data you receive and the data you expect to receive

neographikal, 2020-12-17:
Minimal-modbus python example:

```
#!/usr/bin/env python3
import minimalmodbus
import serial
import time

instrument = minimalmodbus.Instrument('COM4', 0, debug=True)  # port name, slave address (in decimal)
instrument.mode = minimalmodbus.MODE_RTU
instrument.serial.baudrate = 9600         # Baud
instrument.serial.bytesize = 8
instrument.serial.parity   = serial.PARITY_NONE
instrument.serial.stopbits = 1
instrument.address = 0

while True:
    try:
        instrument.write_registers(0x01, [8, 8, 8, 8])
    except:
        print("Error")
    time.sleep(2)
```

This would try to write 4 consecutive registers starting from 0x01 as per the modbus spec: https://www.simplymodbus.ca/FC16.htm. The implementation calls for function code 16, writing multiple registers while writing only one register with 4 values for 4 devices. This seems quite strange to me.

bobosch, 2020-12-18:
The behavior is defined in the program, not in the library. The program has to decide which register is valid and which not. Yes, register 0x0* are implemented in a proprietary way. I will change the documentation, these registers are not meant to used from external.

Please try the register 0xA0 and higher.

neographikal, 2020-12-18:
Will do, still, if you are doing in a proprietary, why not use the correct function code: write single register. In effect you are writing 32bits to 1 register, uncommon but not against spec (there are more devices doing that).

bobosch, 2021-01-24:
Partially done with commit ad18506c522805bdc812d8aac3ab90fc26d51130
But we decided to use another address as broadcast address to avoid unexpected behavior of some modbus devices.
I will also clean up the current broadcast in the future.

closed 2021-01-24 by bobosch