SmartEVSE/smartevse #44: how can i build the qt4 files

qasimali-byte, 2021-03-19:


mstegen, 2021-03-19:
What do you mean by this question? The project can be build in MPLABX.

qasimali-byte, 2021-03-19:
Well actually I want to ask how can I  compile AN1310 bootloader into qt version .

AndrejValand, 2021-03-20:
You may find the information you seek in the following web sites:
https://www.microchip.com/forums/m633482.aspx
https://github.com/martonmiklos/AN1310-Qt5

qasimali-byte, 2021-03-20:
> You may find the information you seek in the following web sites:
> https://www.microchip.com/forums/m633482.aspx
> https://github.com/martonmiklos/AN1310-Qt5

Thanks, @AndrejValand I tried https://github.com/martonmiklos/AN1310-Qt5. but this program has an issue. when I erased the flash AN1310 never stop erasing.

qasimali-byte, 2021-03-20:
@mstegen @AndrejValand this program needs to upgrade to qt5 version.

qasimali-byte, 2021-03-20:
@mstegen @AndrejValand can you help me with this program https://github.com/martonmiklos/AN1310-Qt5.

AndrejValand, 2021-03-20:
Sorry I do not have the in-depth experience and knowledge of the current state of the AN1310-Qt5 project. You will probably find someone that can help you on the AN1310-Qt5 GitHub repository. I think your issue is not with SmartEVSE.