SmartEVSE/smartevse #45: Strange behaviour

rvdgaag, 2022-02-28:
Smart-evse V2 with 2.3 firmware

When set to smart mode, with both a SDM-630 for the EV and Mains, the charger starts and after a couple of seconds it closes the contactor again, starts again over and over. There are no other heavy users in the home grid active so the full 3x25A is available.

With normal charging everything works fine.

Any ideas ?

mstegen, 2022-02-28:
Did you set two different modbus addresses for each SDM630?

rvdgaag, 2022-02-28:
Yes I did, no 1 is the evse, 11 is the sdm630 for the EV and 12 is for the mains including PV.

mstegen, 2022-02-28:
address 11 and 12 should be ok.
The measured (Mains) currents for each phase should be visible on the LCD of the SmartEVSE. Also when the EV is not charging.
Do these currents look OK?

rvdgaag, 2022-02-28:
Yes it is fine, it worked fine for a couple of months but now it is a disaster.

Also reading the MODBUS registers with an external Iobroker gateway works fine.

rvdgaag, 2022-02-28:
In the mean time I found that when I disconnect the iobroker gateway (which polls data every 10seconds), the Smart EVSE works as it should. To me it looks like the Smart EVSE misses data from the mains meter by collisions ?? And shuts off.

Any suggestions how to solve this ? I bought the ease mainly because I could monitor it via modbus.

bobosch, 2022-02-28:
SmartEVSE interrupts measurement when detecting another modbus master
https://github.com/SmartEVSE/SmartEVSE-2/commit/1f0bc9ffe44c02f13dfd7752a36b76c7062dc0f0
10seconds is quite often, maybe try 30 or more seconds

rvdgaag, 2022-03-04:
Ok topic can be closed, the to frequent data requests caused collisions so the smart else did not receive the data from the SDM-630 and cut off the power.

Thanks for the support

closed 2024-07-11 by mstegen