SmartEVSE/smartevse #46: Can't get the thing charging :)

Hansi3, 2022-04-16:
Hi, I'm new here and trying to charge my car on solar power with Smart EVSE. 
I wired a cabinet with a 4 pole switch, kwh meter, contactor and EVSE controller according to the schedule provided. The thing is wired to an external socket with locking pin (±10m cable trajectory). I also added a LED pushbutton.

Anyhow, I think I configured the menu in a sensible way and the display tells me 'ready to charge'. 
When I plug in a cable there happens exactly nothing. No error, no charging, no socket locking whatsoever.
I already removed the push button cabling to extract a fault possibility. 

Anyone any suggestion on how to debug this? Thanks a lot. Hans from NL

mstegen, 2022-04-16:
A distance of 10 meters from SmartEVSE to socket is on the long side.. especially if you then connect another 5 meters of charging cable. The SmartEVSE was designed to be close to the socket. The sensorbox on the other hand, can be far away, as it is connected with a RS485 connection. distances >100m are no problem.

But in order to troubleshoot this, you can try to measure 12V between the CP and PE/ground connection on the charging socket. There should be no car connected.
If there is no 12V present, the ground connection is most likely not connected to the SmartEVSE's  PE terminal.

I hope this helps. 
-Michael Stegen

Hansi3, 2022-04-16:
I measured .021V over CP/PE on both the socket and the controller. Surprisingly before powering on the whole thing I measured .110V on the same position.

Hansi3, 2022-04-16:
I will check the PE wiring in a while

Hansi3, 2022-04-23:
> A distance of 10 meters from SmartEVSE to socket is on the long side.. especially if you then connect another 5 meters of charging cable. The SmartEVSE was designed to be close to the socket. The [...]

Hi Michael,
My last reply wasn't correct. In a hurry I had my multimeter on AC (stupid..). There is a DC voltage of 11.8V over the CP/PE. I checked the wiring and can't find a fault. Do I have any alternative besides dismantling the entire setup and replacing it with a setup with controller and socket in one cabinet? Thanks, Hans

mstegen, 2022-04-23:
Some other things to check. You mentioned you are using a charging socket with locking pin.

Maybe obvious, but i have seen it before:
- make sure the locking pin is not preventing you from plugging in the charging cable all the way into the socket.
- plug in the charging cable into the socket, and measure the CP/PE again on the charging plug that normally would plug into the EV.
- if you do not measure 12V here, the charging socket is most likely mounted on a thick plate, which prevents you from fully inserting the cable into the socket.

I hope this helps.

Hansi3, 2022-04-23:
> 
Altough obvious I never thought of it Michael. You're help is appreciated a lot. There's no voltage on the plugged in cable, most likely since there's a substantial panel thickness. I'll take it apart and test with the socket removed from the panel. I'll post an update. I really appreciate your swift helpful posts. Regards, Hans

Hansi3, 2022-04-25:
Done! It works. Turns out the problem was in the locking solenoid that was wired reversed.

closed 2022-04-25 by Hansi3

mstegen, 2022-04-25:
I'm glad you figured out the problem, and it's fixed now!